meta-flutter/meta-flutter #510: A lot of performance issue without 3D GPU (G2D)

fleroyener, 2024-07-01:
Hello,

I'm currently using a SOM based on IMX9352 but this chip doesn't have a 3D GPU integrated. 
I'm based on kirkstone version and flutter : 3.19.0. (using wayland & weston compositor) 

I see that i have a lot of performance issue (lagging). 
Before enabling G2D on weston i had also issue with weston-terminal but now with G2D enabled i only have performance issue with flutter application. 

Is it possible to disable 3D rendering and use only G2D with flutter-client ?

Don't hesitate if you need further information,
Thank's in advance.

jwinarske, 2024-07-02:
@fleroyener Software backend is not currently supported/implemented in ivi-homescreen.  Weston does support software rasterization.  You might try Vulkan backend with swiftshader and see if things improve.

https://github.com/jwinarske/meta-vulkan/blob/kirkstone/recipes-graphics/swiftshader/swiftshader_git.bb

jwinarske, 2024-09-28:
@fleroyener Does the IMX9352 have a DRM frame buffer implementation?

jwinarske, 2025-11-10:
If it supports a DRM frame buffer you can likely use flutter-pi; I have confirmed this works on a beagle bone black.

This will also be supported in ivi-homescreen v3.