meta-flutter/meta-flutter #511: flutter-auto: Building issue with Vulkan backend

fleroyener, 2024-07-02:
Hi,

I'm trying to compile flutter-auto v2 with yocto kirkstone branch with Vulkan feature enabled. 

But i encountered issue with this compilation : 

| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:37:9: note: expanded from macro 'CHECK_VK_RESULT'
|     vk::resultCheck(static_cast<vk::Result>(x), LOCATION); \
|     ~~~~^
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:502:3: error: no member named 'resultCheck' in namespace 'vk'
|   CHECK_VK_RESULT(d.vkGetPhysicalDeviceSurfacePresentModesKHR(
|   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:37:9: note: expanded from macro 'CHECK_VK_RESULT'
|     vk::resultCheck(static_cast<vk::Result>(x), LOCATION); \
|     ~~~~^
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:505:3: error: no member named 'resultCheck' in namespace 'vk'
|   CHECK_VK_RESULT(d.vkGetPhysicalDeviceSurfacePresentModesKHR(
|   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:37:9: note: expanded from macro 'CHECK_VK_RESULT'
|     vk::resultCheck(static_cast<vk::Result>(x), LOCATION); \
|     ~~~~^
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:544:3: error: no member named 'resultCheck' in namespace 'vk'
|   CHECK_VK_RESULT(result);
|   ^~~~~~~~~~~~~~~~~~~~~~~
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:37:9: note: expanded from macro 'CHECK_VK_RESULT'
|     vk::resultCheck(static_cast<vk::Result>(x), LOCATION); \
|     ~~~~^
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:554:3: error: no member named 'resultCheck' in namespace 'vk'
|   CHECK_VK_RESULT(
|   ^~~~~~~~~~~~~~~~
| /home/debiancompile/msc-ldk/build/01057/tmp/work/cortexa55-poky-linux/flutter-auto/2.0-r0/git/shell/backend/wayland_vulkan/wayland_vulkan.cc:37:9: note: expanded from macro 'CHECK_VK_RESULT'
|     vk::resultCheck(static_cast<vk::Result>(x), LOCATION); \
|     ~~~~^
| fatal error: too many errors emitted, stopping now [-ferror-limit=]

Could you help with this ?

Thank's in advance,
Florent LEROY.

jwinarske, 2024-07-02:
@fleroyener I haven't pushed cherry-picks to kirkstone yet.  Only scarthgap is building to spec.  You can try cherry picking today; might just work.  I won't be looking at that for another week

jwinarske, 2024-09-10:
Will be resolved with https://github.com/meta-flutter/meta-flutter/issues/538

jwinarske, 2024-12-21:
Closing this since fixes went in some time back.  Feel free to reopen if needed

closed 2024-12-21 by jwinarske