meta-flutter/meta-flutter #520: flutter-engine do_fetch fails.

Davidcorb, 2024-07-09:
I am compartively new in using flutter for my scarthgap project and while building flutter-engine it fails with fetch error as follows:

do_fetch.log has the following last lines:
[
failed to generate venv derivation: failed to build package: python_venv-2g70ubjrirt8f3itthk50q0mfk: failed to execute builder: exit status 1

ERROR: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'gn://github.com/flutter/engine.git;gn_name=src/flutter')
DEBUG: Python function base_do_fetch finished
DEBUG: Python function do_fetch finished
]
Did browse through couple of closed issue and understood that gn:// is a gclient and something needs to be added to local.conf to resolve this issue but was not able to excatly get what should be added to local.conf to build gn:// client or just flutter-engine.

It would be really great if anyone can help up

jwinarske, 2024-07-09:
@Davidcorb Hi there.  It could be a variety of things.  

I have seen issues related to fetching the flutter engine from time to time (not related to meta-flutter); specifically when working on workspace-automation.

If it persists you might try `bitbake flutter-engine -cdo_cleansstate`, then `bitbake flutter-engine`

I haven't seen any recent issues (past couple of weeks); haven't built in ~4 days.

Scarthgap references:
https://github.com/meta-flutter/meta-flutter/issues/450
https://github.com/meta-flutter/meta-flutter/issues/461
https://github.com/meta-flutter/meta-flutter/issues/512

Davidcorb, 2024-07-10:
Hi @jwinarske,

Thank-you for the reply.

I tried building flutter-engine as single package and it passed. (bitbake flutter-engine)

But the moment i try to build with it core-image-minimal it fails with the above error. tried cleaning and builiding it again but same issue.

I feel the issue somewhere down the line is that fetching takes too much of a time ~10+mins (maybe it even includes the time fetcher is built) and that is where it timeouts with other execution threads being active.

Do you think that could be an issue ? Can something be done to increase the fetcher speed or speed up the gn:// client ?
 
P.S- My network is good and cloning the repo for flutter-engine seperately just took 30 seconds so network might not be an issue.

jwinarske, 2024-07-10:
@Davidcorb how many HW threads does your build machine have and what's the internet speed/bandwidth?  
My two build machines have 24 and 48 HW threads.  I have also successfully ran on a Mac M2 in UTM/Ubuntu.

Davidcorb, 2024-07-15:
@jwinarske sorry for delay in getting back to you. 
My build machine has 48 HW threads ana a bandwith of 1GBPS speed.
OS is Ubuntu 22.04.

jwinarske, 2024-08-07:
@Davidcorb Where are you at with this issue?

Have you correctly added meta-flutter and meta-flutter-apps?

The only conf/local.conf change would be to remove debug and profile package config values; production release or speed up build time

Davidcorb, 2024-09-17:
Yes, the layers were added correctly and no debug or profile pkgs were a part of the build. Had no success with even more bulky machines but to no avail

Well this took time and we were on debug and workaround mode. we somewhere had a success with a workaround last night

We actually dropped the meta-flutter layer and created own flutter recipe file which downloads only the required version of flutter-engine rather then the entire repo.  The difference in download is pretty much big with repo sizing around 800mbs to just the single version sizing 23mb. Replaced gn:// fecther with normal git fetcher and a few more twiks to get going.

jwinarske, 2024-09-17:
@Davidcorb not clear where the trouble is.  Many people are actively using this layer.

I have pre-builts published as releases here for tooling:
https://github.com/meta-flutter/flutter-engine

jwinarske, 2024-09-28:
Closing since issue is not directly being addressed.

closed 2024-09-28 by jwinarske