meta-flutter/meta-flutter #548: yaml python dependency breaks containerized builds

StaticRocket, 2024-08-15:
This layer added a host dependency on the `yaml` python module. This is not yet installed by default in the [crops/poky](https://github.com/crops/poky-container/pkgs/container/poky#poky-container) build containers.

jwinarske, 2024-08-15:
Yep.  It needs the python3-yaml-native added to the path.  Working on it

jwinarske, 2024-08-15:
I think all it needs is below in common.inc; need confirmation
```
inherit python3-dir
```

jwinarske, 2024-08-15:
@StaticRocket  See if this helps: https://github.com/meta-flutter/meta-flutter/pull/549

jwinarske, 2024-08-15:
@StaticRocket building `scarthgap` with this container I'm hitting

```
ERROR: re2c-native-3.1-r0 do_configure: configure failed
...
| configure: error: no suitable Python interpreter found
```

jwinarske, 2024-08-15:
got past it.  Building with dummy kernel image

jwinarske, 2024-08-15:
I have a repo.  Working on fix

jwinarske, 2024-08-15:
I have it to this:
```
Package         Version
--------------- -------
build           1.1.1
Cython          3.0.8
flit_core       3.9.0
installer       0.7.0
packaging       23.2
pip             24.0
pycurl          7.45.2
pyproject_hooks 1.0.0
PyYAML          6.0.1
setuptools      69.1.1
wheel           0.42.0
```
right before the `import pycurl` fails

`PYTHONVERBOSE` doesn't provide any clues

jwinarske, 2024-08-15:
```
     0533:    """
     0534:    run_command('python3 -m pip list', os.getcwd())
     0535:
     0536:    import io
 *** 0537:    import pycurl
     0538:    from urllib.parse import quote
     0539:
     0540:    if url == '':
     0541:        bb.warn(f'Missing url in description, using default')
Exception: ModuleNotFoundError: No module named 'pycurl'
```

jwinarske, 2024-08-15:
I'm going to swap wget for pycurl.  It may take a few days

closed 2024-09-03 by jwinarske