meta-flutter/meta-flutter #550: Which layer provides python3-pycurl-native? Now flutter app build depends on this

JesvinJoseph, 2024-08-15:
I am getting the below error while building flutter gallery

`ERROR: Nothing PROVIDES 'python3-pycurl-native' (but meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-gallery_2.10.2.bb DEPENDS on or otherwise requires it). Close matches:
  python3-ply-native
  python3-py-native
  python3-pycairo-native
ERROR: Required build target 'flutter-gallery' has no buildable providers.
Missing or unbuildable dependency chain was: ['flutter-gallery', 'python3-pycurl-native']
`

jwinarske, 2024-08-15:
https://github.com/meta-flutter/meta-flutter/pull/551

meta-python is located under meta-openembedded

JesvinJoseph, 2024-08-16:
@jwinarske ,

Thank you for the response. Now getting the error

> No module named 'pycurl'

closed 2024-08-16 by jwinarske

jwinarske, 2024-08-16:
@JesvinJoseph The short term workaround is to install `pycurl` on the build host; unless you're fixed on a container that doesn't have it installed like the TI guy.