meta-flutter/meta-flutter #556: flutter-sdk `do_archive_pub_cache task`

jwinarske, 2024-08-21:
Flutter SDK needs a `do_archive_cache_task`.

This would walk the SDK tree archiving all pubspec.lock files; pre-populating the $DL_DIR/pub_cache folder.

This alleviates `n` number of application recipes downloading the same packages at once.

Another approach would be to skip the file if it's already locked.

JesvinJoseph, 2024-08-22:
> ModuleNotFoundError: No module named 'yaml'

jwinarske, 2024-08-22:
@JesvinJoseph
> ModuleNotFoundError: No module named 'yaml'

How is this related to adding a new task to the flutter-sdk recipe?

JesvinJoseph, 2024-08-23:
> @JesvinJoseph
> 
> > ModuleNotFoundError: No module named 'yaml'
> 
> How is this related to adding a new task to the flutter-sdk recipe?

Not with this task. I mean after merging the branches, I can see this error while building the app. I will post it in another topic

closed 2024-09-03 by jwinarske