meta-flutter/meta-flutter #557: do_restore_pub_cache fails on scarthegap for any flutter-app

pogojotz, 2024-08-29:
I just updated from same Flutter 3.19-ish version to your current 3.24.1 on Scarthgap branch. Now I cannot build my flutter-app anymore. I tested with flutter-gallery with the same result:

<details>
<summary>cat build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/temp/log.do_restore_pub_cache.15275</summary>

```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/work/build/../src/poky/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/meta-flutter/recipes-graphics/flutter-sdk/flutter-sdk_git.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/meta-openembedded/meta-python/recipes-devtools/python/python3-pycurl_7.45.2.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/work/build/../src/poky/meta/recipes-devtools/python/python3-pyyaml_6.0.1.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['gettext-minimal-native', 'cmake-native', 'libtool-native', 'm4-native', 'quilt-native', 'texinfo-dummy-native', 'flutter-sdk-native', 'python3-pycurl-native', 'openssl-native', 'dbus-native', 'expat-native', 'gettext-native', 'glib-2.0-native', 'libxml2-native', 'ncurses-native', 'util-linux-libuuid-native', 'util-linux-native', 'zlib-native', 'flex-native', 'gnu-config-native', 'libedit-native', 'make-native', 'patch-native', 'perl-native', 'python3-build-native', 'python3-cython-native', 'python3-flit-core-native', 'python3-installer-native', 'python3-packaging-native', 'python3-pyproject-hooks-native', 'python3-pyyaml-native', 'python3-setuptools-native', 'python3-wheel-native', 'python3-native', 'bzip2-native', 'libnsl2-native', 'libtirpc-native', 'unzip-native', 'xz-native', 'zstd-native', 'gdk-pixbuf-native', 'gobject-introspection-native', 'gtk+3-native', 'cairo-native', 'fontconfig-native', 'freetype-native', 'harfbuzz-native', 'libjpeg-turbo-native', 'pango-native', 'libpthread-stubs-native', 'libx11-native', 'libxau-native', 'libxcb-native', 'libxcomposite-native', 'libxcursor-native', 'libxdamage-native', 'libxdmcp-native', 'libxext-native', 'libxfixes-native', 'libxft-native', 'libxi-native', 'libxrandr-native', 'libxrender-native', 'libxtst-native', 'pixman-native', 'xtrans-native', 'xcb-proto-native', 'xorgproto-native', 'util-macros-native', 'libpng-native', 'at-spi2-core-native', 'attr-native', 'ca-certificates-native', 'curl-native', 'fribidi-native', 'gdbm-native', 'icu-native', 'itstool-native', 'libcap-ng-native', 'libffi-native', 'libpcre2-native', 'libyaml-native', 'shared-mime-info-native', 'sqlite3-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_restore_pub_cache
DEBUG: environment: environ({'includedir': '/usr/include', 'STRINGS': 'aarch64-taclos-linux-strings', 'PYTHON_LIBRARY': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/lib/libpython3.12.so', 'BUILD_LD': 'ld ', 'infodir': '/usr/share/info', 'mandir': '/usr/share/man', 'BUILD_NM': 'nm', 'sbindir': '/usr/sbin', 'docdir': '/usr/share/doc', 'BUILD_LDFLAGS': '-L/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/usr/lib                         -L/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/lib                         -Wl,-rpath,/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/work/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -pthread', 'PKG_CONFIG_SYSTEM_INCLUDE_PATH': '/usr/include', 'PKG_CONFIG_DIR': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/lib/pkgconfig', 'localstatedir': '/var', 'libdir': '/usr/lib', 'BUILD_CPPFLAGS': '-isystem/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native/usr/include', 'CFLAGS': ' -O2 -pipe -g -feliminate-unused-debug-types -fcanon-prefix-map  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot=  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot=  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native= ', 'bindir': '/usr/bin', 'CPP': 'aarch64-taclos-linux-gcc -E --sysroot=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot  -mcpu=cortex-a72+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security', 'CXXFLAGS': ' -O2 -pipe -g -feliminate-unused-debug-types -fcanon-prefix-map  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot=  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot=  -fdebug-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot-native=  -fvisibility-inlines-hidden', 'prefix': '/usr', 'PKG_CONFIG_SYSTEM_LIBRARY_PATH': '/usr/lib:/usr/lib', 'BUILD_RANLIB': 'ranlib -D', 'nonarch_base_libdir': '/usr/lib', 'NM': 'aarch64-taclos-linux-gcc-nm', 'PYTHONHASHSEED': '0', 'datadir': '/usr/share', 'BUILD_CC': 'gcc ', 'LD': 'aarch64-taclos-linux-ld --sysroot=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot ', 'exec_prefix': '/usr', 'PYTHON_INCLUDE_DIR': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/include/python3.12', 'HOME': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2', 'BUILD_STRIP': 'strip', 'base_sbindir': '/usr/sbin', 'SOURCE_DATE_EPOCH': '1718290435', 'LC_ALL': 'en_US.UTF-8', 'GIT_CEILING_DIRECTORIES': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2', 'PSEUDO_DISABLED': '1', 'base_prefix': '', 'PKG_CONFIG_PATH': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/lib/pkgconfig:/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/share/pkgconfig:/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/lib/pkgconfig:/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/share/pkgconfig', 'systemd_unitdir': '/usr/lib/systemd', 'STAGING_INCDIR': '/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/recipe-sysroot/usr/include', 'base_libdir': '/usr/lib', 'AS': 'aarch64-taclos-linux-as ', 'servicedir': '/srv', 'systemd_user_unitdir': '/usr/lib/systemd/user', 'LDFLAGS': '-Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fcanon-prefix-map  -fmacro-prefix-map=/work/build/tmp/work/cortexa72-taclos-linux/flutter-gallery/2.10.2/git=/usr/src/debug/flutter-gallery/2.10.2  -fdebug-prefix-map=/work/build/tmp/work/corte [... cut ...]
     0795:    shutil.copy(str(file), hosted_cache_path)
     0796:
     0797:    #
Exception: FileNotFoundError

DEBUG: Python function do_restore_pub_cache finished

```
</details>

<details>
<summary>My docker image, I build in, is a modified crops/poky:</summary>

```
FROM crops/poky:ubuntu-22.04

USER root

RUN echo -e "\nHost *\n    StrictHostKeyChecking no" >> /etc/ssh/ssh_config

RUN apt update && \
    apt install -y \
        nano \
        vim \
        python3-yaml

USER usersetup
```
</details>

Am I missing something?

pogojotz, 2024-08-30:
I just finished a bitbake run with https://github.com/meta-flutter/meta-flutter/commit/05072656476999f1e8955808936623b1ffce18e3 the commit just before https://github.com/meta-flutter/meta-flutter/commit/6b164f504521c4b446c033d13398e607e5a34cd3 and that worked fine.

closed 2024-09-03 by jwinarske

jwinarske, 2024-09-04:
@pogojotz Please try with scarthgap tip of tree.  You should be good to go

pogojotz, 2024-09-05:
Thank you. Will do next week.