meta-flutter/meta-flutter #566: [scarthgap] reproducible builds

jwinarske, 2024-09-10:
This is an umbrella issue for reproducible build issues on `scarthgap`

- [ ] AOT generation.  https://github.com/dart-lang/sdk/issues/55282
- [ ] flutter-sdk-native in Yocto Toolchain

cc @g-scott-murray

camaronut, 2026-04-17:
Hi Joel,

I've been on a 2+ year tour of the bowels of Android, hopefully my employer will give me more time to come back to this now that the trip is winding itself down.

A colleague and I were updating our sources from Dunfell to Scarthgap.  Our company's build flow is typically "do Yocto release of BSP and SDK, give to app developers, repeat".  The app developers then take that SDK and build a flutter front-end and a C++ back-end app, linked together with protobuf.  The app is provisioned to the device separate from the Yocto generated filesystem.  App developers use the pub-cache folder that exists inside the Yocto generated SDK.

We have a .bbappend file in our layer for flutter-sdk_git that adds a step to attempt to bolt-down protoc:
```
do_compile[network] = "1"

do_compile:append() {
	# Use a cache path inside the SDK staging area, not $HOME
	export PUB_CACHE="${S}/.pub-cache"
	# The 25.0.0 is intentional - it bolts down the version
	${S}/bin/dart pub global activate protoc_plugin 25.0.0
}
```
We may move this to do_unpack, that's still ongoing.

We were thinking about this today, the SDK's .pub-cache folder may not be reproducible... should that be added to the list here?

We also found this from googling:
https://pub.dev/packages/pub2yocto

It seems pub2yocto could perhaps be used ahead-of-time to help write the SRC_URIs of the dart packages/plugins from a given dart/flutter setup, and that could perhaps be used to feed an SDK build recipe.  Unfortunately this also sounds like a maintenance burden to make updates.

If you have any creative ideas on how to improve the pub-cache handling inside the SDK, we're all ears.
Thanks!

jwinarske, 2026-08-29:
The **AOT generation** item here splits into two independent leaks. Both are now tracked:

- **App `libapp.so`** — #819, fixed. The generated Dart plugin registrant was named to the frontend by absolute path, and that path became the library's import URI, which the AOT image retains. It was the *only* absolute path in a release image, since everything else is reached through `--packages` and carries a `package:` URI. Landed on wrynose (#820), master (#821) and scarthgap (#822). Verified two ways: replaying the recipe's own `do_compile` commands reproduces the shipped `libapp.so` byte for byte and goes from 1 TMPDIR reference to 0, and scarthgap CI went from 3+3 `buildpaths` issues on those two apps to none, with `do_compile` and `do_package_qa` confirmed to have re-run.

- **`dart-sdk`'s `gen_kernel_aot.dart.snapshot`** — #827, **still open**. The other 4 of that run's 10 `buildpaths` issues. Not addressed by #819, and worth noting that master and wrynose only appear clean because their `dart-sdk` recipe carries `buildpaths` in `INSANE_SKIP:${PN}` while scarthgap does not — a divergence introduced incidentally in the 3.47.1 roll.

The linked [dart-lang/sdk#55282](https://github.com/dart-lang/sdk/issues/55282) is a real issue but was not the cause of the app-side warning; that one was fixable here.

jwinarske, 2026-08-30:
Status on the **AOT generation** item — it was two independent leaks, and they have separated cleanly.

**App `libapp.so` — done.** The generated Dart plugin registrant was named to the frontend by absolute path, and that path became the library's import URI, which the AOT image retains. It was the only absolute path in a release image, since everything else is reached through `--packages` and carries a `package:` URI. Fixed on all five branches (#819: #820, #821, #822, #837, #838).

Verified rather than assumed: replaying a recipe's own `do_compile` commands reproduced the shipped `libapp.so` byte for byte and went from 1 TMPDIR reference to 0, and scarthgap CI went from the 3+3 `buildpaths` issues quoted at the top of this issue to none, with `do_package_qa` confirmed to have re-run rather than been skipped.

The `buildpaths` check is also enabled again for app packages — it had been switched off in `flutter-app.inc`, which is why this shipped unnoticed for so long and why scarthgap was the only branch ever reporting it (#828, #829, #830).

**`dart-sdk` snapshots — upstream.** The other 4 of the 10 issues in that run. `bin/snapshots/*.dart.snapshot` each embed their entry point's URI, because those entry points live in `pkg/<name>/bin/` rather than `lib/` and so have no `package:` URI. Not ours to fix: the rule is `sdk/utils/application_snapshot.gni` and the artifacts are built by Dart's own builders — 14 of 14 snapshots in the *released* SDK carry `/b/s/w/ir/x/w/sdk/...`. Filed as dart-lang/sdk#64169, tracked here as #827.

**On the reference to dart-lang/sdk#55282 at the top of this issue:** that one was closed as a Flutter problem and moved to flutter/flutter#165111, which is still open and is exactly the registrant leak above. Our fix and its verification are now posted there, since it applies to `flutter build` generally and not only to Yocto.

jwinarske, 2026-08-30:
Progress on this umbrella, and an audit of the four items against current state.

| item | state |
|---|---|
| AOT generation | **app half done** (#819, all five branches); dart-sdk half is #827, now upstream as dart-lang/sdk#64169 |
| flutter-sdk-native in Yocto Toolchain | **capability exists** — `BBCLASSEXTEND = "native nativesdk"`, and the README documents `TOOLCHAIN_HOST_TASK:append = " nativesdk-flutter-sdk"`. Worth confirming whether this checkbox meant something beyond that |
| gn fetch premirror | **done** — PR #269 was closed unmerged, but the work landed by another route; README's "Mirroring gn-fetched sources" documents `PREMIRRORS`, `MIRRORS`, `BB_FETCH_PREMIRRORONLY` and the two non-obvious details |
| reproducible engine builds | open, tracked as #271 |

## Side finding: dead buildpaths suppressions

Auditing item 2 turned up `INSANE_SKIP:class-nativesdk += "buildpaths"` on flutter-sdk, so I enumerated every `buildpaths` suppression in the layer. The `-dbg` ones on the v3 recipes were **dead**: `DEBUG_PREFIX_MAP` is in `TARGET_CFLAGS` by default and rewrites the build directory out of DWARF, so there was nothing to suppress. Verified against real packages — three `-dbg` packages, zero files containing TMPDIR, and `DW_AT_comp_dir = /usr/src/debug/glibc/2.44+git/csu` rather than a WORKDIR path. oe-core carries no `-dbg` buildpaths skip anywhere across its tree.

Removed on wrynose (#841, CI green with the check fatal), master (#846) and scarthgap (#847).

## Still suppressed, deliberately

| suppression | why it stays |
|---|---|
| `dart-sdk` `${PN}` | real leak, upstream — #827 |
| `flutter-engine` `${PN}-test` | not built in any tree I could inspect |
| `flutter-sdk` `class-nativesdk` | same |
| `flutter-auto_2.0`, `ivi-homescreen_2.0` `-dbg` | different source series; the evidence above does not cover them |

The last four are removable on the same reasoning as the v3 pair, but I would rather check them against real packages than infer it — inferring is how the dead ones got there.

## What is left for this umbrella

#271 is the only remaining item that is ours to do. #827 needs upstream to move.

jwinarske, 2026-09-09:
The `dart-sdk` snapshot leak tracked in #827 is fixed on master as of #956 --
a backport of [dart CL 543800](https://dart-review.googlesource.com/c/sdk/+/543800)
for dart-lang/sdk#64169, which stops the builder's absolute path being baked
into everything under `bin/snapshots/`.

Verified by removing `buildpaths` from the recipe's `INSANE_SKIP` in the same
change, so the QA check ran rather than being suppressed: four machines
green, no `buildpaths` warnings.

That was the second item under this umbrella, after #819. Release branches
still to follow.

jwinarske, 2026-09-09:
#827 is closed -- the `dart-sdk` snapshot leak is fixed on all five branches
(#956, #957, #958, #959, #960), by backporting
[dart CL 543800](https://dart-review.googlesource.com/c/sdk/+/543800) rather
than suppressing the warning.

No branch suppresses `buildpaths` for `dart-sdk` any more, and the three that
never did have stopped emitting it.

That is the second item under this umbrella after #819. Still open here: the
engine's own executables, which is a different mechanism -- prefix-map
coverage in the gn build rather than a dart snapshot URI.

jwinarske, 2026-09-26:
Walking the four items against current state. All four are resolved.

| item | state |
|---|---|
| AOT generation | done. App `libapp.so` was #819 (all five branches); the `dart-sdk` snapshot half was #827, fixed by backporting [dart CL 543800](https://dart-review.googlesource.com/c/sdk/+/543800) rather than suppressing it (#956-#960) |
| flutter-sdk-native in Yocto Toolchain | done. `BBCLASSEXTEND = "native nativesdk"` on the recipe, `TOOLCHAIN_HOST_TASK:append = " nativesdk-flutter-sdk"` documented in the README. Since #1009 the engine host tools and `gen_snapshot` are staged the same way, through `flutter-engine-sdk` and `dart-sdk-tools`, so the SDK side of this is wider than it was when the box was written |
| gn fetch premirror | done. PR #269 was closed unmerged but the work landed by another route; README documents `PREMIRRORS`, `MIRRORS` and `BB_FETCH_PREMIRRORONLY` |
| reproducible engine builds | done, #271 closed. 35 of 35 packaged files byte-identical across two machines with equal build-ids, after fixing a clock-dependent `engine_version` SHA the fetcher was baking into the engine (#1063 and its ports) |

The check that mattered here: scarthgap CI emits zero `buildpaths` on all four machines as of the last run, and the packaged engine has no TMPDIR references in release or debug. `buildpaths` is WARN_QA on scarthgap, so it was worth confirming nothing was quietly still leaking under a warning nobody reads.

Closing. #271 carries the engine reproducibility work, and anything new gets its own issue rather than another box here.

closed 2026-09-26 by jwinarske