meta-flutter/meta-flutter #568: Kirkstone with Flutter 3.24.2

blindmatrix, 2024-09-12:
Hi everyone!

It looks like there are plans to update the Kirkstone branch to align more closely with Scarthgap. However, I needed Flutter 3.24.2 today, so I cherry-picked several configuration files (most of them) from Scarthgap to get this working.

If you're interested, you can merge from this branch: https://github.com/blindmatrix/meta-flutter/tree/kirkstone-3.24.2. I understand if this doesn't align with #538.

But if it does, this branch worked well for my scenario. :)

blindmatrix, 2024-09-13:
@jwinarske Not sure if the thumbs up was a "yes" or a show of interest, so I made a pull request, https://github.com/meta-flutter/meta-flutter/pull/569

jwinarske, 2024-09-13:
Merged

closed 2024-09-13 by jwinarske