meta-flutter/meta-flutter #571: Issue building Flutter gallery on Yocto (Toradex Apalis i.MX8, kirkstone branch)

hareendranmg, 2024-09-20:
Hello,

I'm new to Flutter on Yocto and am trying to build an image for my Toradex Apalis i.MX8 using the latest kirkstone branch. However, I'm encountering an error when running the following command to build the image, which includes the flutter-gallery package:

```bash
PARALLEL_MAKE="-j `nproc`" BB_NUMBER_THREADS="`nproc`" bitbake tdx-reference-minimal-image
```

```vbnet
WARNING: You have included the meta-tpm layer, but 'tpm or tpm2' has not been enabled in your DISTRO_FEATURES. Some bbappend files and preferred version setting may not take effect. See the meta-tpm README for details on enabling tpm support.
Loading cache: 100% |####################################################################################################################################################| Time: 0:00:00
Loaded 4823 entries from dependency cache.
WARNING: No recipes in default available for:
  /home/hareendran/oe-core-flutter_new/build/../layers/meta-toradex-nxp/recipes-bsp/imx-mkimage/imx-mkimage_1.0.bbappend
  /home/hareendran/oe-core-flutter_new/build/../layers/meta-toradex-nxp/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.20.0.imx.bbappend
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'flutter-gallery' (but /home/hareendran/oe-core-flutter_new/build/../layers/meta-toradex-demos/recipes-images/images/tdx-reference-minimal-image.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'flutter-gallery' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['flutter-gallery']
ERROR: Required build target 'tdx-reference-minimal-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['tdx-reference-minimal-image', 'flutter-gallery']

Summary: There were 2 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.

```

It seems like the flutter-gallery package is unbuildable or missing. I've the following packages are included in my local.conf:
```bash
FLUTTER_SDK_TAG = "3.24.2"
CORE_IMAGE_EXTRA_INSTALL += "flutter-engine"
CORE_IMAGE_EXTRA_INSTALL += "flutter-sdk"
CORE_IMAGE_EXTRA_INSTALL += "flutter-pi"
CORE_IMAGE_EXTRA_INSTALL += "packagegroup-flutter-sdk-deps"
CORE_IMAGE_EXTRA_INSTALL += "flutter-wayland-client"
CORE_IMAGE_EXTRA_INSTALL += "flutter-gallery"

```

**The image builds successfully without the flutter-gallery package.**

Markus43, 2024-09-23:
It got a new name and (I believe new) repository location:
https://github.com/meta-flutter/meta-flutter/blob/scarthgap/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-dev-integration-tests-flutter-gallery_git.bb

I believe quite a lot of people searched for this recipe after the recent update ... 

Would it perhaps make sense to define one specific package group with the most common flutter apps like e.g. the old and the new gallery app, and guarantee to keep the group's recipe name for users of this layer? Most people who include this layer in a development image will want a small fallback set of apps to try after upgrading or when the own app breaks. packagegroup-flutter is a bit too big for that IMHO ...

jwinarske, 2024-09-26:
@hareendranmg The Flutter Gallery repo has been archived as indicated by it's repo:  https://github.com/flutter/gallery

As @Markus43 points out, it is available under the Flutter SDK integration tests.  Unclear if it works the same.  I opted to start using the Flutter Sample material_3_demo app.

jwinarske, 2024-09-26:
@Markus43 I use package groups for integration testing; it's far easier than including each app independently.

jwinarske, 2024-09-26:
@hareendranmg `packagegroup-flutter-sdk-deps` is for use case where Yocto SDK image includes the Flutter SDK.  It's not a requirement otherwise, unless you want to develop directly on the target; host side development is a much better experience.

jwinarske, 2024-10-17:
Closing due to inactivity from user.  User can re-open as needed.

closed 2024-10-17 by jwinarske