meta-flutter/meta-flutter #572: rive-text-0.3.2: undefined symbol

julianscheel, 2024-09-26:
I'm building rive-text on kirkstone, but get the following error trying to LD_PRELOAD it at runtime:

flutter-client: symbol lookup error: /usr/lib64/librive_text.so.0.3.2: undefined symbol: _ZN8facebook4yoga5Event7publishERK6YGNodeNS1_4TypeERKNS1_4DataE

Any thoughts what's causing it?

jwinarske, 2024-09-26:
The rive-common package uses `DynamicLibrary.process()` to acquire symbols:
https://github.com/meta-flutter/rive-common/blob/main/lib/src/dynamic_library_helper.dart#L49

Alternatives to LD_PRELOAD which work:
* link the shared module (librive_text.so) to the embedder executable
* dlopen using `RTLD_NOW|RTLD_GLOBAL` flags.  Example: https://github.com/toyota-connected/ivi-homescreen-plugins/blob/v2.0/plugins/rive_text/librive_text.cc#L42

I can imagine if you were to build rive-text using a specific sysroot, then attempt try to run on a different sysroot you might see this issue.  What do you see when you run `ldd /usr/lib64/librive_text.so.0.3.2` on the device in question?

julianscheel, 2024-09-27:
ldd is fine:
```
# LD_TRACE_LOADED_OBJECTS=1 /lib/ld-linux-aarch64.so.1 /usr/lib64/librive_text.so.0.3.2
	linux-vdso.so.1 (0x0000ffff989b5000)
	libc++.so.1 => /usr/lib64/libc++.so.1 (0x0000ffff985e0000)
	libc++abi.so.1 => /usr/lib64/libc++abi.so.1 (0x0000ffff98590000)
	libm.so.6 => /lib64/libm.so.6 (0x0000ffff984f0000)
	libc.so.6 => /lib64/libc.so.6 (0x0000ffff98340000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff9897c000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x0000ffff98310000)
```

I'm not using it on a different sysroot than the build rootfs.
I think the symbol should come from the Yoga library, which librive_text should have inbuilt?

julianscheel, 2024-09-27:
I've updated to 0.4.11, which fixes the issue.

closed 2024-09-27 by julianscheel