meta-flutter/meta-flutter #574: Build failed with musl libc

andryblack, 2024-10-04:
Building failed with TCLIBC = "musl" configured. Need add check.

jwinarske, 2024-10-04:
Yep musl support is not present at this time.  If you can provide SoC/BSP layers you are using, I can take a look at it in a could of weeks when I'm back from holidays.

It expands an already very big test matrix.

andryblack, 2024-10-04:
openembedded-core/meta \
meta-openembedded/meta-oe \
meta-raspberrypi \
meta-clang \
meta-flutter \
meta-openembedded/meta-python \
meta-openembedded/meta-networking \
meta-openembedded/meta-multimedia \
meta-flutter/meta-flutter-apps \

target machine raspberrypi4

jwinarske, 2024-10-05:
Okay, so building flutter-engine with clang external toolchain, and setting `TCLIBC = "musl" ` at top level. 

I suspect if you override the clang layer flutter engine bbappend (clear set values) for flutter-engine, the engine would build.  Your initial issue doesn't indicate what doesn't build; that might be helpful to include as meta-flutter has many recipes.

What embedder are you trying to use on the RPI4?

jwinarske, 2024-10-06:
A key fact missing from this issue is what branch are you using.  Please provide.

andryblack, 2024-10-06:
Primary issue flutter-engine uses its own stale libcxx, with predefined config. Work now on solution.

andryblack, 2024-10-06:
scarthgap branch

closed 2024-10-14 by jwinarske