meta-flutter/meta-flutter #575: Unable to build on a linux arm64 host

Simone-RM, 2024-10-04:
I'm trying to compile a flutter app on a linux arm64 host (using the kirkstone branch).

Compile fails when invoking:

```
NOTE: Running [...build/tmp/work/armv8a-poky-linux/myapp/git+.../recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/dart-sdk/bin/dartaotruntime --disable-analytics --disable-dart-dev ...build/tmp/work/armv8a-poky-linux/myapp/git+.../recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/linux-arm64/frontend_server_aot.dart.snapshot --sdk-root ...build/tmp/work/armv8a-poky-linux/myapp/git+.../recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/common/flutter_patched_sdk_product/ --target=flutter -Ddart.vm.profile=false -Ddart.vm.product=true --aot --tfa --target-os linux --packages ...build/tmp/work/armv8a-poky-linux/myapp/git+.../git//.dart_tool/package_config.json --output-dill ...build/tmp/work/armv8a-poky-linux/myapp/git+.../git//.dart_tool/flutter_build/0b9b511f3403fbcc9d48f37b9714668c/app.dill --depfile ...build/tmp/work/armv8a-poky-linux/myapp/git+.../git//.dart_tool/flutter_build/0b9b511f3403fbcc9d48f37b9714668c/kernel_snapshot_program.d --source file://...build/tmp/work/armv8a-poky-linux/myapp/git+.../git//.dart_tool/flutter_build/dart_plugin_registrant.dart --source package:flutter/src/dart_plugin_registrant.dart -Dflutter.dart_plugin_registrant=file://...build/tmp/work/armv8a-poky-linux/myapp/git+.../git//.dart_tool/flutter_build/dart_plugin_registrant.dart --native-assets ...build/tmp/work/armv8a-poky-linux/myapp/git+.../git//.dart_tool/flutter_build/0b9b511f3403fbcc9d48f37b9714668c/native_assets.yaml --verbosity=error package:myapp/main.dart] in ...build/tmp/work/armv8a-poky-linux/myapp/git+.../git/
NOTE: Output:
  Loading failed: Architecture mismatch.
  VM initialization failed: Invalid vm isolate snapshot seen
```

After some digging, I found this:
```
# file ...build//tmp/work/armv8a-poky-linux/myapp/git+.../recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/linux-arm64/frontend_server_aot.dart.snapshot
...build//tmp/work/armv8a-poky-linux/myapp/git+.../recipe-sysroot-native/usr/share/flutter/sdk/bin/cache/artifacts/engine/linux-arm64/frontend_server_aot.dart.snapshot: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[md5/uuid]=7d7673e2df1d68aa1222110aef3343b6, with debug_info, not stripped
```

I suspect the problem is that snapshot which seems to be in the wrong architecture...

jwinarske, 2024-10-04:
Note there are a couple of other open issues related to arm64 host building.

andryblack, 2024-10-04:
same issue with build with docker image on Mac with M1 processor

jwinarske, 2024-10-05:
Note at this time arm64 host builds are the lowest priority.

Tested PRs are welcome; would require a full test pass for both x86_64 and arm64 host.

andryblack, 2024-10-05:
https://storage.googleapis.com/flutter_infra_release/flutter/a6bd3f1de158bb61090e0c8053df93a10cb548e1/linux-arm64/artifacts.zip
downloaded by flutter sdk include frontend_server_aot.dart.snapshot  for x86-64

jwinarske, 2024-10-06:
If you build on an arm64 host the native host tools in the engine SDK would be arm64.  If you want to run tools on a x64 host, then you have to build on x64 host.

Or fetch engine SDK from here:
https://github.com/meta-flutter/flutter-engine

andryblack, 2024-10-06:
this happened on arm64 host with arm64 target

jwinarske, 2024-10-06:
@andryblack 
> this happened on arm64 host with arm64 target

What meta-flutter branch?  The assumption is you are using `scarthgap`.  If not then move to `scarthgap`.

What are the values of `CLANG_BUILD_ARCH` and `GN_ARGS` in `log.do_configure`?

andryblack, 2024-10-06:
its not flutter-engine issue, but flutter-sdk. error occur on build examples apps, flutter-sdk download [artifacts.zip](https://storage.googleapis.com/flutter_infra_release/flutter/a6bd3f1de158bb61090e0c8053df93a10cb548e1/linux-arm64/artifacts.zip) from google and it is wrong, need use self builded artefacts from flutter-engine: add --build-engine-artifacts gn flag and use produced artifacts.zip for apps build

andryblack, 2024-10-06:
Temporary solution use frontend_server.dart.snapshot from dart-sdk, it is for correct architecture  `frontend_snapshot = f'{flutter_sdk}/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot'`

Simone-RM, 2024-10-07:
I did discover the same workaround (with the aot frontend in my case). I also opened an issue on flutter (https://github.com/flutter/flutter/issues/156309) since I guess it's their bug to fix...

jwinarske, 2024-10-17:
flutter-engine in scarthgap should compile now with and without musl.
Only outstanding item (app build) I'm aware of is https://github.com/meta-flutter/meta-flutter/pull/580

lhoward, 2025-01-16:
My current issue building the engine on arm64 is flutter/issues#157576 (needing to remove OpenJDK from DEPS). I'll try work on a patch for this.

jwinarske, 2025-07-17:
Draft: https://github.com/meta-flutter/meta-flutter/pull/686
Seeing some strange behavior around files not showing up during do_fetch.  It's fetching, but files are not where they should be; first attempt fails, second works.

I was able to build a core-image-weston sdimage.

lhoward, 2025-07-17:
Excellent, I'll test locally as soon as I can.

jwinarske, 2026-08-24:
works in master/wrynose.  Changes are getting back ported

jwinarske, 2026-08-29:
Fixed on every branch.

Your diagnosis was right — that snapshot is the wrong architecture, and the reason is that it is a **host** tool living in a **target**-named directory:

```python
frontend_snapshot = f'{flutter_sdk}/bin/cache/artifacts/engine/linux-{clang_build_arch(d)}/frontend_server_aot.dart.snapshot'
```

`clang_build_arch(d)` names the target, so building for arm64 selects `artifacts/engine/linux-arm64/…`. But the frontend server runs on the build host, and the published engine artifacts under that path are built for x86-64 whatever the directory is called — exactly what your `file` output showed. On an arm64 host `dartaotruntime` is then handed an x86-64 snapshot and refuses it.

The `dart-sdk/bin/snapshots/` copy is host-native, so it is preferred now, with the engine one kept as a fallback for an SDK that does not ship it:

```python
sdk_snapshot      = f'{flutter_sdk}/bin/cache/dart-sdk/bin/snapshots/frontend_server_aot.dart.snapshot'
engine_snapshot   = f'{flutter_sdk}/bin/cache/artifacts/engine/linux-{clang_build_arch(d)}/frontend_server_aot.dart.snapshot'
frontend_snapshot = sdk_snapshot if os.path.exists(sdk_snapshot) else engine_snapshot
```

master and wrynose had this already; it is now on the release branches too — #816 (kirkstone, the branch you reported against), #817 (scarthgap), #818 (dunfell).

## What was and was not verified

CI is green on all four machines for each branch, but **every runner here is x86-64**, where both paths resolve to the same architecture. So those runs demonstrate no regression, and that `dart-sdk/bin/snapshots/frontend_server_aot.dart.snapshot` really is present in the 3.47.1 SDK and is what gets used — they cannot demonstrate the arm64-host case, because we have no arm64 build host.

Closing on the strength of the diagnosis matching your evidence exactly. If it still reproduces on your arm64 host after updating, please reopen with the `file` output of whichever snapshot it picks — that will say immediately whether the preference is taking effect or the SDK layout has moved again.

closed 2026-08-29 by jwinarske