meta-flutter/meta-flutter #590: flutter-engine: do_rm_work failed: some vpython files have no write perm

WAKazuyoshiAkiyama, 2024-10-31:
If `INHERIT += "rm_work"` is specified in conf/local.conf, flutter-engine will cause the error in do_rm_work.

```
...
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/activate.fish': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/easy_install-3.8': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/python3': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/pip': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/python': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/activate_this.py': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/wheel': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/bin/pip3': Permission denied
| /workdir/build/tmp/hosttools/rm: cannot remove 'vpython/822d6b/include/python3.8': Permission denied
```

This is because some vpython files have no write permission like the following.

```
$ ls -l vpython/822d6b/bin -l
total 18084
-r--r--r-- 1 pokyuser docker     2154 10月 31 12:19 activate
-r--r--r-- 1 pokyuser docker     1096 10月 31 12:19 activate.csh
-r--r--r-- 1 pokyuser docker     2250 10月 31 12:19 activate.fish
-r--r--r-- 1 pokyuser docker     1137 10月 31 12:19 activate_this.py
-r-xr-xr-x 1 pokyuser docker      463 10月 31 12:19 easy_install
-r-xr-xr-x 1 pokyuser docker      463 10月 31 12:19 easy_install-3.8
-r-xr-xr-x 1 pokyuser docker      435 10月 31 12:19 pip
-r-xr-xr-x 1 pokyuser docker      435 10月 31 12:19 pip3
-r-xr-xr-x 1 pokyuser docker      435 10月 31 12:19 pip3.8
lrwxrwxrwx 1 pokyuser docker        7 10月 31 12:19 python -> python3
-r-xr-xr-x 1 pokyuser docker 18471128 10月 31 12:19 python3
lrwxrwxrwx 1 pokyuser docker        7 10月 31 12:19 python3.8 -> python3
-r-xr-xr-x 1 pokyuser docker     2552 10月 31 12:19 python-config
-r-xr-xr-x 1 pokyuser docker      442 10月 31 12:19 wheel
```
```

jwinarske, 2024-10-31:
In all my tests today, I had INHERIT += "rm_work" defined in local.conf, for all cases.  I was building with podman and container.

how do I get a repro?

jwinarske, 2024-10-31:
Also what branch?

WAKazuyoshiAkiyama, 2024-11-06:
@jwinarske 
I used the dunfell branch.

WAKazuyoshiAkiyama, 2024-11-08:
The same error happened in `kirkstone` branch also.
I will open a PR to avoid this error.

WAKazuyoshiAkiyama, 2024-11-11:
This issue happened recently, but, we have not updated the revision of meta-flutter for 3+ months.
So, I think this issue may be derived from some Google changes.
Perhaps, there may be some missing factors to be fixed for the long term support, maybe, in depot-tools and gn build tools.

I have seen the similar issue (no write perms for installed files) when I handle a recipe using Bazel.
So, depot-tools/flutter-engine might start to use Bazel to fetch vpython.

WAKazuyoshiAkiyama, 2024-11-11:
Unfortunately, The only way to avoid the issue is to run `chmod` in the case of Bazel.

WAKazuyoshiAkiyama, 2024-11-11:
https://github.com/meta-flutter/meta-flutter/pull/604

jwinarske, 2024-11-12:
@WAKazuyoshiAkiyama

On `scarthgap` I run CI using `INHERIT += "rm_work"` on rootless podman.

If you allow write access to $HOME then it will mask container contamination during build.  Rootless podman does not allow $HOME write access.  If you are not running rootless build, I recommend you switch.

I made a recent change in `scarthgap` which resolves vpython issue (using latest depot_tools); the directory (`VPYTHON_VIRTUALENV_ROOT`) has to be created.  I will investigate further.

jwinarske, 2024-11-16:
@WAKazuyoshiAkiyama Issue should be resolved with https://github.com/meta-flutter/meta-flutter/pull/589.  Please check

closed 2024-11-18 by WAKazuyoshiAkiyama

jwinarske, 2024-11-18:
@WAKazuyoshiAkiyama today dunfell has the same fix

WAKazuyoshiAkiyama, 2024-11-18:
Thank you. I could remove vpython.
The cause is that I use the old revision of depot-tools. (Due to https://github.com/meta-flutter/meta-flutter/issues/497)
My network issue was solved, so, I use the current revision of depot-tools, then, I could solve this issue also.
Sorry.