meta-flutter/meta-flutter #603: feature request: define one stable packagegroup name for demo and development images

Markus43, 2024-11-07:
My SOM module manufacturer provides demo images for their customers, see e.g. here: https://github.com/dh-electronics/kas-dhsom . Their SOM works really well with flutter, and in their demo image they have an option to build with flutter support.

FWIW, I use their i.MX8MP module. Their BSP is based on mainline linux' etnaviv graphic drivers and doesnt depend on nxp's layers. I don't need the patch from https://github.com/meta-flutter/meta-flutter/issues/295 for ivi-homescreen.

In such demos, users also want to test such options. They added flutter-gallery in the past for that, then they changed to packagegroup-flutter-test-apps, then that name was removed from meta-flutter ...

I tried to create a PR here: https://github.com/dh-electronics/meta-dhsom-extras/pull/1/ but my last two propsed changes broke quickly again ... Just like my own development image continuously breaks after pulling in new meta-flutter versions, because recipe names cease to exist.

It would be really nice if meta-flutter-apps could provide at least one package group name that is guaranteed to persist for such use cases. Such a package group could contain only a few basic flutter apps. The old and new gallery app (recently removed without any announcement AFAICS) would be good candidates for such a group IMHO. Apps in that group could change in case of build issues with newer flutter versions.

This could be used to:
* demonstrate that flutter is working in hardware manufacturers' demo images
* include in development images, so users of this layer can quickly check if known-good apps are working when something with their own app recipes stops to work.

jwinarske, 2024-11-07:
@Markus43

Hi Markus

* The flutter-gallery has been archived upstream; no longer supported.  It's available under a different name in the flutter-sdk; if you want to use it.  I have moved to using a recipe from the 1st party flutter sample repo: `flutter-samples-material-3-demo`

* As part of maintenance when moving to a new Flutter SDK, existing projects that lag or have been archived will become broken.  To minimize the time I spend on this layer, I have moved to a model that targets 1st party (track stable releases), and a select group of third party repos.  The primary method I use for moving to a Flutter SDK is auto-rolling.  If things become broken or archived upstream, I remove them as I have little time or interest to come up with patches for repos that are outdated.  The model favors maintained flutter apps.

* There was some churn during the transition to auto-roll support; recipe names.  That has since stabilized as the script is doing it's job; it does the recipe naming.  I don't see this pattern changing in the foreseeable future, as it does what it was designed to do.

* Stability stages are focused around business stakeholders deliverables.  AGL being one, as well as my employer.  If a SOM vendor, or third party agency wants commercial support, wants to fund features, or to simply collaborate (produce contributions opposed to consuming), you/they are welcome to contact me.  PRs are always welcome.  To date I have received no funding for this layer.

* Yes for NXP the issue is specifc to `imx-weston`.  Not all iMX builds use `imx-weston`, which is why I use this wording.

* Prior to any downstream stakeholder release, there will likely be some churn.  If you roll to a commit during active development it will likely present you some challenges.  The best approach here is to ask questions on Discord.

My recommendation is to pick and use an app from meta-flutter-apps first_party folder; such as `flutter-samples-material-3-demo`.

jwinarske, 2024-11-09:
@Markus43 what would be interesting is a functional test app for each embedder.  This would likely be more interesting for a SoM/SoC vendor.

Markus43, 2024-11-12:
> * The flutter-gallery has been archived upstream; no longer supported.  It's available under a different name in the flutter-sdk; if you want to use it.  I have moved to using a recipe from the 1st party flutter sample repo: `flutter-samples-material-3-demo`

Out of interest, why did you remove packagegroup-flutter, together with flutter-dev-integration-tests-flutter-gallery_git.bb and flutter-dev-integration-tests-new-gallery_2.10.2.bb , in this commit: https://github.com/meta-flutter/meta-flutter/commit/9dadce868eebf540a6ea879fb93346df67c73046 ?

> * Stability stages are focused around business stakeholders deliverables.  AGL being one, as well as my employer.  If a SOM vendor, or third party agency wants commercial support, wants to fund [...]

Many thanks for your continued support here! The point that I wanted to make with this issue is to give some feedback regarding how your layer is used. 

> * Prior to any downstream stakeholder release, there will likely be some churn.  If you roll to a commit during active development it will likely present you some challenges.  The best approach here is to ask questions on Discord.

That's what I do when I run into any issues (username FourtyTwo there). By the way, how can a user of this layer identify revisions that have been used for downstream stakeholder releases?
 
> My recommendation is to pick and use an app from meta-flutter-apps first_party folder; such as `flutter-samples-material-3-demo`.

Thanks for the recommendation, I'll use that in my images and suggest my som vendor to use it from now on.

Markus43, 2024-11-12:
> @Markus43 what would be interesting is a functional test app for each embedder. This would likely be more interesting for a SoM/SoC vendor.

That sounds interesting, yes. I would even go one step further and say that some kind of common benchmarks to run against each embedder would be cool. How would you know if an embedder works on a given hardware or if you need a bit more RAM for your usecase? I believe many people make very subjective and biased decisions here.

Something like that could likely be very good advertizing for ivi-homescreen - just saying ;-)

jwinarske, 2024-11-15:
@Markus43 

> Out of interest, why did you remove packagegroup-flutter, together with flutter-dev-integration-tests-flutter-gallery_git.bb and flutter-dev-integration-tests-new-gallery_2.10.2.bb , in this commit: [9dadce8](https://github.com/meta-flutter/meta-flutter/commit/9dadce868eebf540a6ea879fb93346df67c73046) ?

Build errors and lack of time.  The auto-roll script will pull in all the pubspec.yaml in a repo, there are more exceptions (should not build) in there, I opted to remove it due to inadequate time to implement and validate.

> Many thanks for your continued support here! The point that I wanted to make with this issue is to give some feedback regarding how your layer is used.

You're welcome, and I appreciate the thought.  Constructive input is always welcome as it helps to improve things.

> That's what I do when I run into any issues (username FourtyTwo there). By the way, how can a user of this layer identify revisions that have been used for downstream stakeholder releases?

From general end user not an easy way ATM.  Tagging would certainly improve that problem.

> That sounds interesting, yes. I would even go one step further and say that some kind of common benchmarks to run against each embedder would be cool. How would you know if an embedder works on a [...]

My initial plan is to start with a showcase app, that also acts as a functional unit test case for all of the plugins.  It would help to prepare for releases.  Having to run through each functional use case for each plugin is a bit of a pain.

> Something like that could likely be very good advertizing for ivi-homescreen - just saying ;-)

It would make a nice Medium article...