meta-flutter/meta-flutter #605: flatbuffers is missing in vanilla poky build

kwavnet, 2024-11-13:
Setting up a build with just poky and meta-flutter in it, fails with

```
ERROR: Required build target 'meta-world-pkgdata' has no buildable providers.
Missing or unbuildable dependency chain was: ['meta-world-pkgdata', 'firebase-cpp-sdk', 'flatbuffers']
```

`flatbuffers` is served by meta-oe

I see two options here

- make meta-oe (aka openembedded-layer) mandatory
- move everything that invokes flatbuffers (from a quick check that would be firebase and therefore flutter-auto) to a dynamic-layer

What would be the preferred option? (I'm willing to post a patch)

jwinarske, 2024-11-14:
Hi @kwavnet,

glib, libxml, dbus, etc are in `meta-oe`.  I never called it out as a hard dependency, as `meta-oe` is required to build a generic linux image.

I'll accept a PR on `scarthgap` with `meta-oe` as a required layer if you like.

kwavnet, 2024-11-27:
Looks like the LAYERDEPENDS is already set correctly - I think that was happening to me on kirkstone, potentially not with the latest - just double checked, on all supported branches the dependencies are set.

Consider this fixed

closed 2024-11-27 by kwavnet