meta-flutter/meta-flutter #608: gn fetcher fails for noexec download dir

thochstein, 2024-11-19:
We are doing a VM build with a shared download directory on a noexec mount. The gn fetcher does try to execute dart from the download directory, which results in the following error:

```
PermissionError: [Errno 13] Permission denied: '/efs/shared/yocto-dl-dir/gn/https__github.com_flutter_engine.git/src/flutter/third_party/dart/tools/sdks/dart-sdk/bin/dart'
```

This is a big problem for us, as our shared directory can only be noexec.

I believe this usage is at best non-standard (it's the first time we've run across such an error in our builds), and perhaps a security hole. Is it possible to rework this so dart is executed from the standard recipe-sysroot-native instead?

jwinarske, 2024-11-19:
@thochstein This happens in "run-hooks" phase, which happens after the engine source tree is fetched.  It executes from the source repo.  See:
https://github.com/flutter/engine/blob/main/DEPS#L939-L1089
Any of the items without a condition always run, and the ones with a condition run based on the variable value.  This is what I set that affects the hooks:
```
GN_CUSTOM_VARS ?= '\
{\
    "download_android_deps": False, \
    "download_windows_deps": False, \
    "download_linux_deps": False,   \
}'
```
So any of the hooks with these conditions, will not run.  It narrows down what actually runs.

You can try adding this to your local.conf:
```
EXTRA_GN_SYNC:pn-flutter-engine = "--no-verify --shallow --no-history -R -D"
```
It's supposed to disable hooks.

This ran before?  If so with what meta-flutter branch/commit.  Perhaps the delta is `--no-verify`.  If so, then when it does run (invoked by configure), it would run in the recipes $WORKDIR.  It would then likely require network access which breaks LTS.

I would expect your setup to have issues with all of the gn based recipes in the layer.

Regarding security concerns I build on RedHat (selinux enabled) using rootless Podman.  When something does execute it only has user access of which the container runs in, not root.  I lock down the runner user.  Running non-rootless Docker it has root access.  Running non-rootless Docker/Podman is a serious security concern.  The CI jobs reference these containers:  https://github.com/meta-flutter/docker-containers.  Getting the rootless setup on a selinux system is not that simple.

jwinarske, 2024-11-19:
This function would require modification for the `gn fetcher`:
https://github.com/meta-flutter/meta-flutter/blob/scarthgap/lib/gn.py#L34-L88

These for the `cip fetcher`:
https://github.com/meta-flutter/meta-flutter/blob/scarthgap/lib/cipd.py#L58-L89
https://github.com/meta-flutter/meta-flutter/blob/scarthgap/lib/cipd.py#L58-L89

thochstein, 2024-11-19:
> You can try adding this to your local.conf:
> 
> ```
> EXTRA_GN_SYNC:pn-flutter-engine = "--no-verify --shallow --no-history -R -D"
> ```

```
]';             gclient sync --force --no-verify --shallow --no-history -R -D --revision db49896cf25ceabc44096d5f088d86414e05a7aa -j 1 -v failed with exit code 2, output:
mkdir: missing operand
Try 'mkdir --help' for more information.
WARNING: Your metrics.cfg file was invalid or nonexistent. A new one will be created.
Usage: gclient.py sync [options]

gclient.py: error: no such option: --no-verify
```

Note the `-j 1` was a debugging attempt. Please ignore.

thochstein, 2024-11-19:
`gclient.py: error: no such option: --no-hooks`

jwinarske, 2024-11-20:
What does gclient —help output?

thochstein, 2024-11-21:
Oh, realized I can use `-c devshell`.

```
root@yb20:~/styhead/build/tmp/work/armv8a-poky-linux/flutter-engine/3.24.4/src# ../recipe-sysroot-native/usr/share/depot_tools/gclient --help
Usage: gclient.py <command> [options]

Meta checkout dependency manager for Git.

Commands are:
  config       creates a .gclient file in the current directory
  diff         displays local diff for every dependencies
  fetch        fetches upstream commits for all modules
  flatten      flattens the solutions into a single DEPS file
  getdep       gets revision information and variable values from a DEPS file
  gitmodules   adds or updates Git Submodules based on the contents of the DEPS file
  grep         greps through git repos managed by gclient
  help         prints list of commands or help for a specific command
  installhooks installs gclient git hooks
  metrics      reports, and optionally modifies, the status of metric collection
  pack         generates a patch which can be applied at the root of the tree
  recurse      operates [command args ...] on all the dependencies
  revert       reverts all modifications in every dependencies
  revinfo      outputs revision info mapping for the client and its dependencies
  root         outputs the solution root (or current dir if there isn't one)
  runhooks     runs hooks for files that have been modified in the local working copy
  setdep       modifies dependency revisions and variable values in a DEPS file
  status       shows modification status for every dependencies
  sync         checkout/update all modules
  validate     validates the .gclient and DEPS syntax
  verify       verifies the DEPS file deps are only from allowed_hosts

Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  -j JOBS, --jobs=JOBS  Specify how many SCM commands can run in parallel;
                        defaults to 96 on this machine
  -v, --verbose         Produces additional output for diagnostics. Can be
                        used up to three times for more logging info.
  --gclientfile=CONFIG_FILENAME
                        Specify an alternate .gclient file
  --spec=SPEC           create a gclient file containing the provided string.
                        Due to Cygwin/Python brokenness, it can't contain any
                        newlines.
  --no-nag-max          Ignored for backwards compatibility.
```

I tried `--version` as well.

```
root@yb20:~/styhead/build/tmp/work/armv8a-poky-linux/flutter-engine/3.24.4/src# ../recipe-sysroot-native/usr/share/depot_tools/gclient --version
Your copy of depot_tools is configured to fetch from an obsolete URL:

  https://chromium.googlesource.com/chromium/tools/depot_tools

OK to update it to https://chromium.googlesource.com/chromium/tools/depot_tools.git ? [Y/n] n
./cipd: line 143: ../recipe-sysroot-native/usr/share/depot_tools/cipd_client_version.digests: No such file or directory
Platform linux-amd64 is not supported by the CIPD client bootstrap: there's no pinned SHA256 hash for it in the *.digests file.
../recipe-sysroot-native/usr/share/depot_tools/bootstrap_python3: line 32: bootstrap-2@3.11.8.chromium.35_bin/python3/bin/python3: No such file or directory
```

jwinarske, 2024-11-22:
Looking at a recent version these are the current options
```
$ gclient sync --help
Usage: gclient.py sync [options]

Checkout/update all modules.

Options:
  --version             show program's version number and exit
  -h, --help            show this help message and exit
  -j JOBS, --jobs=JOBS  Specify how many SCM commands can run in parallel;
                        defaults to 32 on this machine
  -v, --verbose         Produces additional output for diagnostics. Can be
                        used up to three times for more logging info.
  --gclientfile=CONFIG_FILENAME
                        Specify an alternate .gclient file
  --spec=SPEC           create a gclient file containing the provided string.
                        Due to Cygwin/Python brokenness, it can't contain any
                        newlines.
  --no-nag-max          Ignored for backwards compatibility.
  -f, --force           force update even for unchanged modules
  -n, --nohooks         don't run hooks after the update is complete
  -p, --noprehooks      don't run pre-DEPS hooks
  -r REV, --revision=REV
                        Enforces git ref/hash for the solutions with the
                        format src@rev. The src@ part is optional and can be
                        skipped. You can also specify URLs instead of paths
                        and gclient will find the solution corresponding to
                        the given URL. If a path is also specified, the URL
                        takes precedence. -r can be used multiple times when
                        .gclient has multiple solutions configured, and will
                        work even if the src@ part is skipped. Revision
                        numbers (e.g. 31000 or r31000) are not supported.
  --patch-ref=GERRIT_REF
                        Patches the given reference with the format
                        dep@target-ref:patch-ref. For |dep|, you can specify
                        URLs as well as paths, with URLs taking preference.
                        |patch-ref| will be applied to |dep|, rebased on top
                        of what |dep| was synced to, and a soft reset will be
                        done. Use --no-rebase-patch-ref and --no-reset-patch-
                        ref to disable this behavior. |target-ref| is the
                        target branch against which a patch was created, it is
                        used to determine which commits from the |patch-ref|
                        actually constitute a patch.
  -t, --download-topics
                        Downloads and patches locally changes from all open
                        Gerrit CLs that have the same topic as the changes in
                        the specified patch_refs. Only works if atleast one
                        --patch-ref is specified.
  --with_branch_heads   Clone git "branch_heads" refspecs in addition to the
                        default refspecs. This adds about 1/2GB to a full
                        checkout. (git only)
  --with_tags           Clone git tags in addition to the default refspecs.
  -H, --head            DEPRECATED: only made sense with safesync urls.
  -D, --delete_unversioned_trees
                        Deletes from the working copy any dependencies that
                        have been removed since the last sync, as long as
                        there are no local modifications. When used with
                        --force, such dependencies are removed even if they
                        have local modifications. When used with --reset, all
                        untracked directories are removed from the working
                        copy, excluding those which are explicitly ignored in
                        the repository.
  -R, --reset           resets any local changes before updating (git only)
  -M, --merge           merge upstream changes instead of trying to fast-
                        forward or rebase
  -A, --auto_rebase     Automatically rebase repositories against local
                        checkout during update (git only).
  --deps=OS_LIST        override deps for the specified (comma-separated)
                        platform(s); 'all' will process all deps_os references
[... 28 lines cut ...]
      update files from SCM according to current configuration, for
      all modules (useful for recovering files deleted from local copy)
  gclient sync --revision src@GIT_COMMIT_OR_REF
      update src directory to GIT_COMMIT_OR_REF

JSON output format:
If the --output-json option is specified, the following document structure will
be emitted to the provided file. 'null' entries may occur for subprojects which
are present in the gclient solution, but were not processed (due to custom_deps,
os_deps, etc.)

{
  "solutions" : {
    "<name>": {  # <name> is the posix-normalized path to the solution.
      "revision": [<git id hex string>|null],
      "scm": ["git"|null],
    }
  }
}
```

This is the one to use
```
-n, --nohooks         don't run hooks after the update is complete
```

thochstein, 2024-11-23:
That did get me past the fetcher error. Next problem is in `do_compile`:

```
ERROR: Logfile of failure stored in: /home/bamboo/automation/styhead/fsl-imx-internal-xwayland/yocto/build/tmp/work/armv8a-poky-linux/flutter-engine/3.24.4/temp/log.do_compile.36686
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: FLUTTER_RUNTIME_MODES=debug profile release
| ninja: Entering directory `out/linux_debug_arm64'
| [0/1] Regenerating ninja files
| WARNING at build arg file (use "gn args <out_dir>" to edit):57:19: Build argument has no effect.
| angle_build_all = false
|                   ^----
| The variable "angle_build_all" was set as a build argument
| but never appeared in a declare_args() block in any buildfile.
| 
| To view all possible args, run "gn args --list <out_dir>"
| 
| The build continued as if that argument was unspecified.
| 
| ninja: error: '../../flutter/tools/const_finder/.dart_tool/package_config.json', needed by 'gen/const_finder.dart.snapshot', missing and no known rule to make it
| WARNING: exit code 1 from a shell command.
```

Could this be due to the Styhead `WORKDIR/UNPACKDIR` change?

jwinarske, 2024-11-23:
You need the hooks to run.  To support DL_DIR as a noexec mount you need changes to the code I pointed to prior.

If you plan on shipping product, I would only use an LTS release.  styhead is not an LTS release.

jwinarske, 2025-06-05:
Fix candidate:
https://github.com/meta-flutter/meta-flutter/tree/jw/scarthgap-3.32.0

I’ve refactored fetching of archive into WORKDIR, when hooks can execute.  After archive is created it’s copied to DL_DIR.  All gn specific recipes were affected.

If you have a chance to confirm this branch by Tuesday next week, that would be greatly appreciated.

jwinarske, 2025-06-05:
Scarthgap -> kirkstone -> dunfell, then master.

Scarthgap first due to AGL being based on Scarthgap.

thochstein, 2025-06-05:
Hi @jwinarske, thanks for the fix and the heads up. I'm trying to build it locally first before trying the build machine.

```
ERROR: flutter-packages-shared-preferences-shared-preferences-shared-preferences-tool-1.0.1-r0 do_compile: flutter pub get --enforce-lockfile --offline -v failed: 69
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-packages-shared-preferences-shared-preferences-shared-preferences-tool/1.0.1/temp/log.do_compile.1221119
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-packages-shared-preferences-shared-preferences-shared-preferences-tool_1.0.1.bb:do_compile) failed with exit code '1'
ERROR: flutter-samples-provider-shopper-1.0.0-r0 do_compile: flutter pub get --enforce-lockfile --offline -v failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-samples-provider-shopper/1.0.0/temp/log.do_compile.1221118
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-samples-provider-shopper_1.0.0.bb:do_compile) failed with exit code '1'
ERROR: flutter-samples-veggieseasons-1.2.0-r0 do_compile: flutter pub get --enforce-lockfile --offline -v failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-samples-veggieseasons/1.2.0/temp/log.do_compile.1221116
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-samples-veggieseasons_1.2.0.bb:do_compile) failed with exit code '1'
ERROR: flutter-samples-form-app-1.0.0-r0 do_compile: flutter pub get --enforce-lockfile --offline -v failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-samples-form-app/1.0.0/temp/log.do_compile.1221117
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-samples-form-app_1.0.0.bb:do_compile) failed with exit code '1'
ERROR: googleads-flutter-packages-mediation-gma-mediation-moloco-example-git-r0 do_archive_pub_cache: dart pub --suppress-analytics --directory . get --example failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/googleads-flutter-packages-mediation-gma-mediation-moloco-example/git/temp/log.do_archive_pub_cache.1221115
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/googleads-flutter-packages-mediation-gma-mediation-moloco-example_git.bb:do_archive_pub_cache) failed with exit code '1'
ERROR: flutter-packages-go-router-go-router-examples-3.0.1-r0 do_archive_pub_cache: dart pub --suppress-analytics --directory . get --example failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-packages-go-router-go-router-examples/3.0.1/temp/log.do_archive_pub_cache.1221113
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-packages-go-router-go-router-examples_3.0.1.bb:do_archive_pub_cache) failed with exit code '1'
ERROR: googleads-flutter-samples-admob-banner-example-1.0.0-r0 do_archive_pub_cache: dart pub --suppress-analytics --directory . get --example failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/googleads-flutter-samples-admob-banner-example/1.0.0/temp/log.do_archive_pub_cache.1221114
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/googleads-flutter-samples-admob-banner-example_1.0.0.bb:do_archive_pub_cache) failed with exit code '1'
ERROR: flutter-super-dash-super-dash-1.0.0-r0 do_archive_pub_cache: dart pub --suppress-analytics --directory . get --example failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-super-dash-super-dash/1.0.0/temp/log.do_archive_pub_cache.1221112
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-super-dash-super-dash_1.0.0.bb:do_archive_pub_cache) failed with exit code '1'
ERROR: flutter-packages-google-adsense-google-adsense-example-1.0.0-r0 do_compile: flutter build bundle -v failed: 1
ERROR: Logfile of failure stored in: /home/r60874/walnascar/build/tmp/work/armv8a-poky-linux/flutter-packages-google-adsense-google-adsense-example/1.0.0/temp/log.do_compile.1221139
ERROR: Task (/home/r60874/walnascar/sources/meta-flutter/meta-flutter-apps/recipes-graphics/flutter-apps/first-party/flutter-packages-google-adsense-google-adsense-example_1.0.0.bb:do_compile) failed with exit code '1'
```

This is what my `packagegroup` looks like:

```
inherit packagegroup

RDEPENDS:${PN} = " \
    flutter-wayland-client \
    packagegroup-flutter-packages \
    packagegroup-flutter-samples \
    packagegroup-flutter-super-dash\
    packagegroup-googleads-flutter \
    packagegroup-google-generative-ai-dart"
```

thochstein, 2025-06-06:
Hi @jwinarske, I successfully built your branch using a noexec download dir! Thanks for the fix!

thochstein, 2025-06-09:
Fixed by: #656

closed 2025-07-17 by jwinarske