meta-flutter/meta-flutter #609: Errors on Styhead

thochstein, 2024-11-22:
Trying to build on Styhead and seeing numerous errors like these:

```
ERROR: flutter-samples-testing-app-1.0.0-r0 do_compile: flutter pub get --enforce-lockfile --offline -v failed: 1
ERROR: flutter-super-dash-super-dash-1.0.0-r0 do_archive_pub_cache: dart pub --suppress-analytics --directory . get --example failed: 1
ERROR: flutter-packages-google-sign-in-google-sign-in-google-sign-in-example-git-r0 do_restore_pub_cache: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_restore_pub_cache(d)
     0003:
File: '.../styhead/sources/meta-flutter/conf/include/common.inc', lineno: 295, function: do_restore_pub_cache
     0291:    bb.note(f'Running [{cmd}] in {unpackdir}')
     0292:    ret = subprocess.call(cmd, preexec_fn=subprocess_setup, shell=True, cwd=unpackdir)
     0293:
     0294:    if ret != 0:
 *** 0295:        raise UnpackError(f'Unpack command {cmd} failed with return value {ret}', localpath)
     0296:
     0297:    # restore flutter pub get artifacts
     0298:    cmd = \
     0299:        f'mv .project/.dart_tool {app_root}/ | true; ' \
Exception: bb.fetch2.UnpackError: Unpack failure for URL: '.../styhead/downloads/flutter-pub-cache-flutter-packages-google-sign-in-google-sign-in-google-sign-in-example-ce68d07c590e2f6f044af67d5e8b9ccf-3.24.3.tar.bz2'. Unpack command PATH=".../styhead/sources/poky/scripts:.../styhead/build/tmp/work/armv8a-poky-linux/flutter-packages-google-sign-in-google-sign-in-google-sign-in-example/git/recipe-sysroot-native/usr/bin/aarch64-poky-linux:.../styhead/build/tmp/work/armv8a-poky-linux/flutter-packages-google-sign-in-google-sign-in-google-sign-in-example/git/recipe-sysroot/usr/bin/crossscripts:.../styhead/build/tmp/work/armv8a-poky-linux/flutter-packages-google-sign-in-google-sign-in-google-sign-in-example/git/recipe-sysroot-native/usr/sbin:.../styhead/build/tmp/work/armv8a-poky-linux/flutter-packages-google-sign-in-google-sign-in-google-sign-in-example/git/recipe-sysroot-native/usr/bin:.../styhead/build/tmp/work/armv8a-poky-linux/flutter-packages-google-sign-in-google-sign-in-google-sign-in-example/git/recipe-sysroot-native/sbin:.../styhead/build/tmp/work/armv8a-poky-linux/flutter-packages-google-sign-in-google-sign-in-google-sign-in-example/git/recipe-sysroot-native/bin:.../styhead/sources/poky/bitbake/bin:.../styhead/build/tmp/hosttools" pbzip2 -dc -p64 .../styhead/downloads/flutter-pub-cache-flutter-packages-google-sign-in-google-sign-in-google-sign-in-example-ce68d07c590e2f6f044af67d5e8b9ccf-3.24.3.tar.bz2 | tar x --no-same-owner -f - failed with return value 2
```

Perhaps it has something to do with the [WORKDIR changes](https://docs.yoctoproject.org/dev/singleindex.html#workdir-changes)?

jwinarske, 2024-11-22:
@thochstein `styhead` has a new variable introduced called `UNPACKDIR`.  Master branch needs to be updated, and `UNPACKDIR` support needs to be added.  AGL has requested `master` support.  It's in the backlog

closed 2024-12-21 by jwinarske

thochstein, 2025-01-09:
I'm having `flutter-super-dash-super-dash` task `do_archive_pub_cache` fail:

```
NOTE: Running [dart pub --suppress-analytics --directory . get --example] in /home/r60874/styhead/build/tmp/work/armv8a-poky-linux/flutter-super-dash-super-dash/1.0.0/sources-unpack/git/
NOTE: Output:
  Resolving dependencies...
  Because every version of flame_tiled from git depends on flame ^1.23.0 which depends on ordered_set ^6.0.0, every version of flame_tiled from git requires ordered_set ^6.0.0.
  So, because super_dash depends on both ordered_set ^5.0.2 and flame_tiled from git, version solving failed.

  You can try the following suggestion to make the pubspec resolve:
  * Try upgrading your constraint on ordered_set: dart pub add ordered_set:^6.1.1
ERROR: dart pub --suppress-analytics --directory . get --example failed: 1
```