meta-flutter/meta-flutter #646: scarthgap: additional paths needed in debug package

lhoward, 2025-02-20:
I needed the following changes to get `flutter-engine` building with `dbg-pkgs` in `EXTRA_IMAGE_FEATURES`.

```diff
diff --git a/recipes-graphics/flutter-engine/flutter-engine_git.bb b/recipes-graphics/flutter-engine/flutter-engine_git.bb
index c815614..c9bb743 100644
--- a/recipes-graphics/flutter-engine/flutter-engine_git.bb
+++ b/recipes-graphics/flutter-engine/flutter-engine_git.bb
@@ -322,6 +322,8 @@ FILES:${PN} = "\
 
 FILES:${PN}-dbg = "\
     ${FLUTTER_ENGINE_INSTALL_PREFIX}/*/lib/.debug \
+    /usr/src/debug \
+    ${libdir}/debug \
     "
 
 FILES:${PN}-sdk-dev = "\
```

jwinarske, 2025-02-27:
What are the contents of the dbg package with your change?

lhoward, 2025-02-27:
OK, so there a bunch of headers there, perhaps we don't want them?

[flutter-engine-dbg_3.27.1-r0_cortexa72.ipk.files.txt](https://github.com/user-attachments/files/19017845/flutter-engine-dbg_3.27.1-r0_cortexa72.ipk.files.txt)

jwinarske, 2025-02-28:
of the list I think we would only want these:
```
-rwxr-xr-x root/root 253712240 2025-01-11 15:29 ./usr/lib/debug/usr/lib/libflutter_engine.so.debug
-rwxr-xr-x root/root 178847312 2025-01-11 15:29 ./usr/lib/debug/usr/lib/libimpeller.so.debug
-rwxr-xr-x root/root  78202624 2025-01-11 15:29 ./usr/lib/debug/usr/lib/libpath_ops.so.debug
-rwxr-xr-x root/root  28790456 2025-01-11 15:29 ./usr/lib/debug/usr/lib/libtessellator.so.debug
```

lhoward, 2025-02-28:
OK, I'll update accordingly and file a PR when I have a moment

jwinarske, 2025-02-28:
I'm thinking the debug folder install changed in `scarthgap`.   I'm thinking it should be
```
${libdir}/debug/usr/lib/*.debug
```

lhoward, 2025-02-28:
Hmm, the above works, but without also adding `${TARGET_DBGSRC_DIR}` to `FILES:${PN}-dbg` it still fails with unpackaged files. Oddly adding `rm -rf ${D}${TARGET_DBGSRC_DIR}` in `do_install()` doesn't seem to help. You sure you don't want those headers installed? :)

lhoward, 2025-02-28:
FYI `bitbake.conf` defines

```
TARGET_DBGSRC_DIR ?= "/usr/src/debug/${PN}/${PV}"
```

jwinarske, 2025-02-28:
I never considered using "dbg-pkgs" due to the resultant OS image size.  I suppose it would work fine with small images.  Makes sense.

On my local `scarthgap` build flutter-engine-dbg*.rpm using the default "debug-tweaks" is created package with this:

```
joel@air:/mnt/raid10/yocto/scarthgap/up2/tmp/deploy/rpm/corei7_64$ rpm -qlp flutter-engine-dbg-3.27.4-r0.corei7_64.rpm 
/usr
/usr/share
/usr/share/flutter
/usr/share/flutter/3.27.4
/usr/share/flutter/3.27.4/debug
/usr/share/flutter/3.27.4/debug/lib
/usr/share/flutter/3.27.4/debug/lib/.debug
/usr/share/flutter/3.27.4/debug/lib/.debug/libflutter_engine.so
/usr/share/flutter/3.27.4/debug/lib/.debug/libflutter_linux_gtk.so
/usr/share/flutter/3.27.4/debug/lib/.debug/libimpeller.so
/usr/share/flutter/3.27.4/debug/lib/.debug/libpath_ops.so
/usr/share/flutter/3.27.4/debug/lib/.debug/libtessellator.so
/usr/share/flutter/3.27.4/profile
/usr/share/flutter/3.27.4/profile/lib
/usr/share/flutter/3.27.4/profile/lib/.debug
/usr/share/flutter/3.27.4/profile/lib/.debug/libflutter_engine.so
/usr/share/flutter/3.27.4/profile/lib/.debug/libflutter_linux_gtk.so
/usr/share/flutter/3.27.4/profile/lib/.debug/libimpeller.so
/usr/share/flutter/3.27.4/profile/lib/.debug/libpath_ops.so
/usr/share/flutter/3.27.4/profile/lib/.debug/libtessellator.so
/usr/share/flutter/3.27.4/release
/usr/share/flutter/3.27.4/release/lib
/usr/share/flutter/3.27.4/release/lib/.debug
/usr/share/flutter/3.27.4/release/lib/.debug/libflutter_engine.so
/usr/share/flutter/3.27.4/release/lib/.debug/libflutter_linux_gtk.so
/usr/share/flutter/3.27.4/release/lib/.debug/libimpeller.so
/usr/share/flutter/3.27.4/release/lib/.debug/libpath_ops.so
/usr/share/flutter/3.27.4/release/lib/.debug/libtessellator.so
```

Which is enough for all my use cases; off device analyze coredump for required files, resolve dbg packages required, etc.  Automated triage.

I'll run some local experiments with `core-image-minimal` and `dbg-pkgs`.

jwinarske, 2025-02-28:
Manual build of `core-image-minimal` using:
```
EXTRA_IMAGE_FEATURES ?= "debug-tweaks dbg-pkgs"
```
and
```
IMAGE_INSTALL:append = " \
     flutter-engine \
     "
```
builds fine for me

lhoward, 2025-03-01:
Hmm, ok, I just needed the debug packages to do some profiling, I don't necessarily need this to be fixed properly so I'm happy to close this.

closed 2025-03-01 by lhoward