meta-flutter/meta-flutter #651: ParseError at <path...conf/layer.conf:30: unparsed line: 'addpylib ${LAYERDIR}/lib gn'

thb3893, 2025-03-20:
bitbake fsl-image-gui -c populate_sdk
ERROR: ParseError at /home/thomasb/var-fsl-yocto/sources/meta-flutter/conf/layer.conf:30: unparsed line: 'addpylib ${LAYERDIR}/lib gn'

Cloned the who meta-flutter git repo onto my linux computer, and tried to build the image with the bblayers.conf updated, the image.bb file updated, and I am getting the above parse error. I've tried manually typing out the whole file to avoid hidden characters, and I have even tried deleting the repo and cloning it again. 

Any thoughts?

thb3893, 2025-03-20:
Discovered I had to change the layer.conf file to include the kirkstone version i was working with, still running into errors as always... :)

binogoli, 2025-03-21:
I have the same problem, maybe "addpylib" is not working in my version kirkstone ?

binogoli, 2025-03-21:
Hey bro, I just realized that I forget the switch to the kirkstone  branch, this can solve your problem too, I think
```c
git checkout -b kirkstone origin/kirkstone
```

jwinarske, 2025-03-26:
I’ll take a look tomorrow.  It may have crept in from a master cherry pick.

jwinarske, 2025-03-30:
This didn't make it out of the backlog.  I'll take a closer look at this the week of the 14th when I'm back from holidays.

binogoli, 2025-03-31:
> This didn't make it out of the backlog. I'll take a closer look at this the week of the 14th when I'm back from holidays.

Tks for your supporting, hope you enjoy your holidays : )

binogoli, 2025-04-29:
> This didn't make it out of the backlog. I'll take a closer look at this the week of the 14th when I'm back from holidays.

Hi jwinarske，just wanted to gently check in and see if you had a chance to take a look at this. issue

jwinarske, 2025-05-23:
should be addressed by the 30th (AGL Trout release)

jwinarske, 2025-07-13:
@thb3893 Can you confirm is this is outstanding or resolved?