meta-flutter/meta-flutter #652: Update to 3.29.x

julianscheel, 2025-03-24:
Updating to flutter 3.29 is not possible by running the `tools/roll_meta_flutter.py` as the engine repository was moved. See: https://github.com/flutter/engine/blob/main/README.md#flutter-engine

I assume we'll have to change the git uri, and the `S` variable in the flutter-engine recipe at least.

jwinarske, 2025-03-25:
3.29.0 is a breaking change; flutter/engine mono-repo.

Effectively a path change, with a subtle caveat; the engine version scheme is broken in this release.

The internal/bin/engine commit points to something else.  The script that rolls this file is picking the wrong commit.

The commit should be the same commit as the flutter release.

I haven’t heard back from the engine team on this topic as of yet.  So it’s on the shelf until there is quorum.

julianscheel, 2025-03-26:
Thanks for the details @jwinarske 
Is the issue of the broken version scheme tracked in some upstream ticket?

jwinarske, 2025-03-27:
It’s on an internal discord channel at the moment.  I’ll raise GitHub issue tomorrow.  It’s been a while

jwinarske, 2025-03-30:
I'll take a closer look at this the week of the 14th when I'm back from holidays.

julianscheel, 2025-05-08:
@jwinarske Have you had a chance to take a closer look yet?

julianscheel, 2025-06-12:
@jwinarske Thanks a lot works nicely (tested kirkstone only so far). Closing this issue.