meta-flutter/meta-flutter #653: `do_archive_pub_cache` not always re-run

edrose, 2025-03-25:
We're having an issue where the `do_archive_pub_cache` task is not always being re-run.

We have a recipe for an app that has two versions: a stable version with a set version number and a `git+` version that builds the latest commit. It's this git version that isn't working properly. If we step the commit forwards (or use `${AUTOINC}`) then `do_archive_pub_cache` does nothing. The logs show the following:

```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/ed/kas/build/../oe-core/meta/recipes-devtools/quilt/quilt-native_0.67.bb:do_populate_sysroot', 'virtual:native:/home/ed/kas/build/../meta-flutter/recipes-graphics/flutter-sdk/flutter-sdk_git.bb:do_populate_sysroot', 'virtual:native:/home/ed/kas/build/../oe-core/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/ed/kas/build/../oe-core/meta/recipes-extended/pbzip2/pbzip2_1.1.13.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['gettext-minimal-native', 'cmake-native', 'libtool-native', 'm4-native', 'quilt-native', 'texinfo-dummy-native', 'flutter-sdk-native', 'openssl-native', 'dbus-native', 'expat-native', 'gettext-native', 'glib-2.0-native', 'libxml2-native', 'ncurses-native', 'util-linux-libuuid-native', 'util-linux-native', 'zlib-native', 'flex-native', 'gnu-config-native', 'libedit-native', 'make-native', 'patch-native', 'perl-native', 'python3-build-native', 'python3-flit-core-native', 'python3-installer-native', 'python3-packaging-native', 'python3-pyproject-hooks-native', 'python3-setuptools-native', 'python3-wheel-native', 'python3-native', 'bzip2-native', 'libnsl2-native', 'libtirpc-native', 'pbzip2-native', 'unzip-native', 'xz-native', 'zstd-native', 'gdk-pixbuf-native', 'gobject-introspection-native', 'gtk+3-native', 'cairo-native', 'fontconfig-native', 'freetype-native', 'harfbuzz-native', 'libjpeg-turbo-native', 'pango-native', 'libpthread-stubs-native', 'libx11-native', 'libxau-native', 'libxcb-native', 'libxcomposite-native', 'libxcursor-native', 'libxdamage-native', 'libxdmcp-native', 'libxext-native', 'libxfixes-native', 'libxft-native', 'libxi-native', 'libxrandr-native', 'libxrender-native', 'libxtst-native', 'pixman-native', 'xtrans-native', 'xcb-proto-native', 'xorgproto-native', 'util-macros-native', 'libpng-native', 'at-spi2-core-native', 'attr-native', 'ca-certificates-native', 'curl-native', 'fribidi-native', 'gdbm-native', 'icu-native', 'itstool-native', 'libcap-ng-native', 'libffi-native', 'libpcre2-native', 'shared-mime-info-native', 'sqlite3-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_archive_pub_cache
DEBUG: Python function do_archive_pub_cache finished
```

We're on scarthgap, which changed the way package versions are done. Our recipe contains the following:

```
SRCREV = "d9c2b71530102d5e2c6c4b326e76a0bbb5b0cd3d"
PV = "git"
```

Which should set `PKGV` to be `git+d9c2b71530102...`

So for some reason `do_archive_pub_cache` seems to depend on `PV` (which is static in scarthgap) but not `PKGV` or even the result of `do_patch` for when the source code changes.

I've been doing some digging at my end and am still testing the effect of adding `do_archive_pub_cache[deptask] = "do_patch"` to the class

jwinarske, 2025-03-30:
Hi @edrose.  I'll take a closer look at this the week of the 14th, when I'm back from holidays.

jwinarske, 2025-06-05:
@edrose I’ve been swamped with production work. How did the experiments go with 
    do_archive_pub_cache[deptask] = "do_patch" to the class

I’m planning to roll scarthgap early next week.  I’d like to address this one by end of day Monday.

jwinarske, 2025-07-07:
@edrose `pub_cache_archive` is named via `PN` and md5 of the source directory.  If the source changes the md5 will change, regardless of the `PN` value.  Have you tried instrumenting `get_pub_cache_archive_name` to see if the name is unique each time? 

Related code
```
def md5_update_from_dir(directory, hash):
    from pathlib import Path
    assert Path(directory).is_dir()
    for path in sorted(Path(directory).iterdir(), key=lambda p: str(p).lower()):
        hash.update(path.name.encode())
        if path.is_file():
            with open(path, "rb") as f:
                for chunk in iter(lambda: f.read(4096), b""):
                    hash.update(chunk)
        elif path.is_dir():
            hash = md5_update_from_dir(path, hash)
    return hash

def md5_dir(directory):
    import hashlib
    return md5_update_from_dir(directory, hashlib.md5()).hexdigest()

def get_pub_cache_archive_name(d, app_root):
    """
    Returns unique pub cache archive name
    """
    app_root_md5 = md5_dir(app_root)
    pn = d.getVar('PN')
    flutter_sdk_tag = d.getVar('FLUTTER_SDK_TAG')
    pub_cache_archive = f'flutter-pub-cache-{pn}-{app_root_md5}-{flutter_sdk_tag}.tar.bz2'

    workspace = d.getVar('WORKDIR')
    file = os.path.join(workspace, 'PUB_CACHE_ARCHIVE')
    with open(file, 'w+') as f:
        f.write(pub_cache_archive)

    return pub_cache_archive
```

closed 2026-08-25 by jwinarske