meta-flutter/meta-flutter #664: SDK 3.32.0 + riscv (unmatched) compile error in flutter/third_party/abseil-cpp

dl9pf, 2025-06-11:
We see this:
`[1145/10482] CXX obj/flutter/third_party/abseil-cpp/absl/debugging/stacktrace.stacktrace.o
FAILED: obj/flutter/third_party/abseil-cpp/absl/debugging/stacktrace.stacktrace.o
/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/recipe-sysroot-native/usr/bin/clang++ -fuse-ld=lld -Wno-unused-command-line-argument -MD -MF obj/flutter/third_party/abseil-cpp/absl/debuggin
g/stacktrace.stacktrace.o.d --target=riscv64-unknown-linux-gnu --sysroot /home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/recipe-sysroot -DUSE_OPENSSL=1 -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_S
OURCE -D_LARGEFILE64_SOURCE -D_LIBCPP_DISABLE_AVAILABILITY=1 -D_LIBCPP_DISABLE_VISIBILITY_ANNOTATIONS -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -DNDEBUG -DNVALGRIND -DDYNAMIC_ANNOTATIONS_ENABLED=0 -DABSL_ALLOCATOR_NOTHROW=1 -I
../.. -Igen -I../../flutter/third_party/libcxx/include -I../../flutter/third_party/libcxxabi/include -I../../flutter/build/secondary/flutter/third_party/libcxx/config -I../../flutter/third_party/abseil-cpp -fno-strict-aliasing 
-fstack-protector --param=ssp-buffer-size=8 -fPIC -pipe -pthread -fcolor-diagnostics -fdebug-prefix-map=/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/gn/engine/src/= -no-canonical-prefixe
s -fvisibility=hidden --sysroot=/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/recipe-sysroot -Wstring-conversion -Wnewline-eof -O2 -fno-ident -fdata-sections -ffunction-sections -g0 -Wno-
deprecated-non-prototype -Wno-newline-eof -Wno-typedef-redefinition -Wendif-labels -Werror -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-implicit-int-float
-conversion -Wno-deprecated-copy -Wno-psabi -Wno-deprecated-literal-operator -Wno-unqualified-std-cast-call -Wno-non-c-typedef-for-linkage -Wno-range-loop-construct -Wshorten-64-to-32 -Wimplicit-int-conversion -Wsign-compare -W
sign-conversion -Wtautological-unsigned-zero-compare -fvisibility-inlines-hidden -std=c++17 -fno-rtti -nostdinc++ -nostdinc++ -fvisibility=hidden -fno-exceptions -Wno-newline-eof -Wbool-conversion -Wconstant-conversion -Wenum-c
onversion -Wint-conversion -Wliteral-conversion -Wnon-literal-null-conversion -Wnull-conversion -Wobjc-literal-conversion -Wstring-conversion -Wbitfield-enum-conversion -Wno-unknown-warning-option -Wno-deprecated-builtins -g -f
eliminate-unused-debug-types      -fmacro-prefix-map=/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/gn=/usr/src/debug/flutter-engine/3.32.2     -fdebug-prefix-map=/home/dl9pf/AGL/master/bu
ild-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/gn=/usr/src/debug/flutter-engine/3.32.2     -fmacro-prefix-map=/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/gn=/usr/src/debug
/flutter-engine/3.32.2     -fdebug-prefix-map=/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/gn=/usr/src/debug/flutter-engine/3.32.2     -fdebug-prefix-map=/home/dl9pf/AGL/master/build-flu
tter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/recipe-sysroot=     -fmacro-prefix-map=/home/dl9pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/recipe-sysroot=     -fdebug-prefix-map=/home/dl9
pf/AGL/master/build-flutter/tmp/work/riscv64-agl-linux/flutter-engine/3.32.2/recipe-sysroot-native=       -c ../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc -o obj/flutter/third_party/abseil-cpp/absl/debugging
/stacktrace.stacktrace.o
In file included from ../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc:46:
../../flutter/third_party/abseil-cpp/absl/debugging/internal/stacktrace_riscv-inl.inc:154:20: error: implicit conversion loses integer precision: 'uintptr_t' (aka 'unsigned long') to 'int' [-Werror,-Wshorten-64-to-32]
  154 |         sizes[n] = ComputeStackFrameSize(frame_pointer, next_frame_pointer);
      |                  ~ ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../flutter/third_party/abseil-cpp/absl/debugging/internal/stacktrace_riscv-inl.inc:47:37: error: operand of ? changes signedness: 'long' to 'uintptr_t' (aka 'unsigned long') [-Werror,-Wsign-conversion]
   47 |   return low < high ? high_char_ptr - low_char_ptr : kUnknownFrameSize;
      |   ~~~~~~              ~~~~~~~~~~~~~~^~~~~~~~~~~~~~
../../flutter/third_party/abseil-cpp/absl/debugging/internal/stacktrace_riscv-inl.inc:154:20: note: in instantiation of function template specialization 'ComputeStackFrameSize<void *>' requested here
  154 |         sizes[n] = ComputeStackFrameSize(frame_pointer, next_frame_pointer);
      |                    ^
../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc:124:12: note: in instantiation of function template specialization 'UnwindImpl<false, false>' requested here
  124 |       f = &UnwindImpl<false, false>;
      |            ^
In file included from ../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc:46:
../../flutter/third_party/abseil-cpp/absl/debugging/internal/stacktrace_riscv-inl.inc:154:20: error: implicit conversion loses integer precision: 'uintptr_t' (aka 'unsigned long') to 'int' [-Werror,-Wshorten-64-to-32]
  154 |         sizes[n] = ComputeStackFrameSize(frame_pointer, next_frame_pointer);
      |                  ~ ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc:130:12: note: in instantiation of function template specialization 'UnwindImpl<true, false>' requested here
  130 |       f = &UnwindImpl<true, false>;
      |            ^
In file included from ../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc:46:
../../flutter/third_party/abseil-cpp/absl/debugging/internal/stacktrace_riscv-inl.inc:154:20: error: implicit conversion loses integer precision: 'uintptr_t' (aka 'unsigned long') to 'int' [-Werror,-Wshorten-64-to-32]
  154 |         sizes[n] = ComputeStackFrameSize(frame_pointer, next_frame_pointer);
      |                  ~ ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../flutter/third_party/abseil-cpp/absl/debugging/stacktrace.cc:132:12: note: in instantiation of function template specialization 'UnwindImpl<true, true>' requested here
  132 |       f = &UnwindImpl<true, true>;
      |            ^
4 errors generated.
`

In our builds for the 'unmatched' risc-v board in AGL.  I think we need to update the library. 
It looks like we need https://github.com/abseil/abseil-cpp/commit/a3f7e9866c67ca090995aa638276b8fd8938c6f3    added (maybe amongst others).

jwinarske, 2025-06-12:
Got it.  It needs a patch.  Thanks for checking!

jwinarske, 2025-07-13:
resolved with https://github.com/meta-flutter/meta-flutter/pull/684

closed 2025-07-13 by jwinarske