meta-flutter/meta-flutter #688: No plugin library found after building

Jierkai0027, 2025-08-12:
Hi，
I  use meta-flutter to build flutter samples with `bitbake flutter-samples-desktop-photo-search-fluent-ui` as a example. But I found no relevant plugin library in image. I‘ve no idea where the configuration is wrong. I thought I should get a library with the following structure after building. But actually, I got only libapp.so and libflutter_engine.so.

└── usr
└── share
...
├── lib
│   ├── [libapp.so](http://libapp.so/)
│   ├── libayatana-appindicator3.so.1
...
│   ├── [libreqable_http.so](http://libreqable_http.so/)
│   ├── [libreqable_netbare.so](http://libreqable_netbare.so/)
│   ├── [libtray_manager_plugin.so](http://libtray_manager_plugin.so/)
│   ├── [liburl_launcher_linux_plugin.so](http://liburl_launcher_linux_plugin.so/)
│   └── [libwindow_size_plugin.so](http://libwindow_size_plugin.so/)

jwinarske, 2025-08-15:
@Jierkai0027 By design a flutter app recipe will only create libapp.so.  It’s very much inline with Yocto principles.

You need to create a dedicated recipe for each required native component.

thochstein, 2025-08-15:
Hi @jwinarske. I believe this is _not_ in line with Yocto principles, and regular Yocto users looking to add a flutter app to an image to try it will be surprised when it doesn't work.

Whenever a package is added to the rootfs, it is expected that all dependencies will also be added so that the package will work at runtime. Normally that is handled as you describe with a dedicated recipe for each package build. If the dependency has an actual linkage, then Yocto will automatically add the package to the rootfs. If there's not an actual linkage, as with script packages or dynamically loaded libraries, then it is expected that the recipe will explicitly define the `RDEPENDS`.

https://docs.yoctoproject.org/5.2.1/singleindex.html#automatically-added-runtime-dependencies

Do you have a list of apps which will and won't work out-of-the-box? It's certainly okay for things to be in-progress, but that should be clear to the user.

jwinarske, 2025-08-15:
@thochstein I image your repo is built from Flutter command line (developer convenience), and now you want Yocto to generate a LTS reproducible image?

jwinarske, 2025-08-19:
@thochstein Each Linux embedder has a unique build scheme/tooling:

* Flutter GTK - works, embedded performance is poor
* Sony - Sony appears to no longer be maintaining
* Flutter Pi - KMS DRM only; author has changed jobs
* Toyota - v1 (Wayland only - in production, present in AGL releases)
* Toyota - v2 (Wayland only - present in AGL releases)
* Toyota - v3 (Multi-Platform (all platforms, same code base)

I am the author/owner of the Toyota Embedder, and work for Toyota.  My baseline requirements include high performance, production quality, and LTS.

So again depending on what Linux embedder you use there are different constraints, and results.  Multiple SoC vendors use this layer as-is.

I'm available to discuss requirements/planning.

Cheers,
Joel

jierkai0, 2025-08-21:
@jwinarske We are using Sony's solution now.

With my building, I can only see git code in pub_cache instead of building these plugins and installing the library. As discussed above, maybe I need to create separate recipes to handle these plugins？

However, from my research on the flutter Dart plugin, I found that when building a Flutter app, you only need to declare dependencies in pubspec.yaml, without any additional Yocto configuration.

I'm now trying to build window_size in perspective of recipe which is an important plugin. Is this problem related to the choice of Linux embedder, or is this a Sony issue?

BR,
Erkai

karin027, 2025-08-25:
Hi @thochstein Please help evaluate the feasibility of supporting this approach long term.

Jierkai0027, 2025-08-25:
Hi @thochstein Please help evaluate the feasibility of supporting this approach long term.

jwinarske, 2026-08-24:
reopen with artifact list at a minimum.  Then we can discuss a maintenance commitment.

closed 2026-08-24 by jwinarske