meta-flutter/meta-flutter #691: flutter_embedder.h not found in do_install task

lhoward, 2025-08-15:
Recent versions (I haven't bisected exactly where this broke, but I suspect related to a Flutter version upgrade, or upstream consolidation of repositories?) can't seem to find `flutter_embedder.h`.

```
install: cannot stat '/BuildRoot/build/tmp-glibc/work/cortexa72-oe-linux/flutter-engine/3.32.8/gn/out/linux_release_arm64/flutter_embedder.h': No such file or directory
```

The file is instead in `.../flutter-engine/3.32.8/gn/engine/src/out/linux_release_arm64`.

(Note I am building with 3.32.8 – see #690 – but this error occurred with 3.32.5 as well.)

lhoward, 2025-08-15:
Also see: https://huynguyennovem.medium.com/compiling-flutter-engine-locally-what-changes-after-monorepo-part-iv-857fa1b9f4f9

jwinarske, 2025-08-15:
Do you see this issue in meta-flutter with 3.32.5 (the current supported version)?

lhoward, 2025-08-15:
Sorry, this was an error in a local `bbappend`.

closed 2025-08-15 by lhoward